fkanehiro/hrpsys-base #1266: SequencePlayer内にNULLをstd::cerrで出力している

hsnuhayato, 2019-02-26:
当方はrosbridgeにて、
$ rosservice call /SequencePlayerServiceROSBridge/setTargetPose "rarm" [x,y,z] [r,p,y] time
という操作を行っているですが、
するとこの行におけるframe_nameがNULLになり、std::cerrで出力されます。
https://github.com/fkanehiro/hrpsys-base/blob/37cab04094b22af4b11b176b8b657baaafd5a14e/rtc/SequencePlayer/SequencePlayer.cpp#L640
ubuntuだと出力が途切れるだけでシステムは落ちないが、
QNXだとコアダンプしてしまいrtcdが落ちます。
コアダンプ時のrtcd logが
[setTargetPose] Solveing IK with frameMemory fault (core dumped)
と表示されています。

この行を
```c++
 std::cerr << "[setTargetPose] Solveing IK with frame" <<  (frame_name? frame_name:"none_frame") << ", Error " << m_error_pos << m_error_rot << ", Iteration " << m_iteration << std::endl;
```
と直したいですが、いかがでしょうか。
良かったら後ほどpull requestをお送りします。
よろしくお願いします。

ishiguroJSK, 2019-02-26:
僕はあまり使わない機能なので責任が持てませんが，
確かに
```
rtmlaunch hrpsys_ros_bridge_tutorials samplerobot.launch
rosservice call /SequencePlayerServiceROSBridge/setTargetPose "rarm" [0,0,0] [0,0,0] 5.0
rosservice call /SequencePlayerServiceROSBridge/setTargetPose "rarm" [0,0,0] [0,0,0] 5.0
rosservice call /SequencePlayerServiceROSBridge/setTargetPose "rarm" [0,0,0] [0,0,0] 5.0
```
とかやると二回目以降，制御がすぐ帰ってきて挙動がおかしいので，普通にバグだと思います．
```
[ INFO] [1551177549.922333317, 12.456000000]: SequencePlayerServiceROSBridge::setTargetPose() called
[setTargetPose] Solveing IK with frame[ INFO] [1551177549.923991040, 12.458000000]: SequencePlayerServiceROSBridge::setTargetPose() succeeded
[ INFO] [1551177550.407137713, 12.940000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 501[Hz] (exec 1760 Hz, dropped 0)
[ INFO] [1551177551.409470944, 13.942000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 501[Hz] (exec 1769 Hz, dropped 0)
[ INFO] [1551177551.719477545, 14.252000000]: SequencePlayerServiceROSBridge::setTargetPose() called
[ INFO] [1551177551.720400080, 14.254000000]: SequencePlayerServiceROSBridge::setTargetPose() succeeded
[rtmlaunch] check connection/activation
[ INFO] [1551177552.410021462, 14.944000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 501[Hz] (exec 1769 Hz, dropped 0)
[INFO] [WallTime: 1551177552.491191] [15.028000] check 6 collision status, 1.193792 Hz
[ INFO] [1551177552.692453385, 15.226000000]: SequencePlayerServiceROSBridge::setTargetPose() called
[ INFO] [1551177552.693276459, 15.228000000]: SequencePlayerServiceROSBridge::setTargetPose() succeeded
```
確かにSequencePlayerService_impl.cppの
```
CORBA::Boolean SequencePlayerService_impl::setTargetPose(const char* gname, const dSequence& xyz, const dSequence& rpy, CORBA::Double tm){
    char* frame_name = (char *)strrchr(gname, ':');
    if ( frame_name ) {
        ((char *)gname)[frame_name - gname] = '\0'; // cut frame_name, gname[strpos(':')] = 0x00
        frame_name++; // skip ":"
    }
    return m_player->setTargetPose(gname, xyz.get_buffer(), rpy.get_buffer(), tm, frame_name);
}
```
を見るに，"rarm:WAIST"とかではなく"rarm"とだけやるとconst char* frame_nameにNULLが入って再現するので，提案されている修正でPR送っていい気がします．

(本質的にはポインタの多用という問題がある気もしますが・・・)

k-okada, 2019-02-28:
ちゃんと記憶していないので間違ったおとを言っているかもしれないですが，ですが，`frame_name == NULL` というのはそもそも想定していないので，表示の部分の変更ではなくて，その前でエラーとして返すか，デフォルト値(`base_parent_name`?)をセットしてあげるのが良いのではないでしょうか？

hsnuhayato, 2019-02-28:
この辺読んだら、
frame_nameが与えた時に入力のtarget poseがこのframe_nameから見たposeで、
frame_nameがNULLの時に入力のtarget poseが絶対座標系からみたposeになります。
https://github.com/fkanehiro/hrpsys-base/blob/37cab04094b22af4b11b176b8b657baaafd5a14e/rtc/SequencePlayer/SequencePlayer.cpp#L627-L637

例えばnextageにおいて、WAIST座標系=絶対座標系の場合だと、
```bash
$rosservice call /SequencePlayerServiceROSBridge/setTargetPose "rarm:WAIST" [x,y,z] [r,p,y] 1
```
と 
```bash
$rosservice call /SequencePlayerServiceROSBridge/setTargetPose "rarm" [x,y,z] [r,p,y] 1
```
が同じ場所に移動しますので、frame_nameはオプションで与えなくていいと思います。

>デフォルト値(`base_parent_name`?)をセットしてあげるのが良いのではないでしょうか？

nextageみたいなbaseが固定しているロボットだとデフォルトでframe_name = m_robot->rootLink()->nameでいいと思いますが、移動ロボットの場合だとbaseからみたposeと絶対座標系からみたposeが違う可能性があるのではないでしょうか。
設計者の意図を読み間違うかもしれないが、よろしくお願いします。

k-okada, 2019-02-28:
なるほど，おっしゃるとおりです．
まずはこのバグを発見できるように
https://github.com/fkanehiro/hrpsys-base/blob/37cab04094b22af4b11b176b8b657baaafd5a14e/test/test-samplerobot-sequence.py
から呼びだされている．
https://github.com/fkanehiro/hrpsys-base/blob/37cab04094b22af4b11b176b8b657baaafd5a14e/sample/SampleRobot/samplerobot_sequence_player.py#L462-L465
あたりにテストケースを追加してもらえると助かります．

hsnuhayato, 2019-03-01:
テスト追加してみました。
https://github.com/fkanehiro/hrpsys-base/pull/1267
ですが、ubuntuの環境だと自分の修正を入れなくてもこの追加テストは通ります。
修正前だとターミナルのcerr表示が途中で途切れるが、これをテストケースでチェックする方法ご存じでしたら教えて頂けませんでしょうか。
よろしくお願いします。

k-okada, 2019-03-01:
@ishiguroJSK 
> とかやると二回目以降，制御がすぐ帰ってきて挙動がおかしいので，普通にバグだと思います．

とあるので、このおかしい挙動を捕まえられるとよいと思ったのですが。。。

ishiguroJSK, 2019-03-03:
すいません，ちゃんとIK解ける姿勢で確認すると，端末には(seqのcerr出力が？)出なくなるが，seqは動作し続けているようです．シミュレータ上でも動きます．
```
rosservice call /SequencePlayerServiceROSBridge/setTargetPose "larm" [0.1,0.2,1] [-0.0,-1.29,-0.59] 5.0
operation_return: True
rosservice call /SequencePlayerServiceROSBridge/setTargetPose "larm" [0.2,0.2,1] [-0.0,-1.29,-0.59] 5.0
operation_return: True
rosservice call /SequencePlayerServiceROSBridge/setTargetPose "larm" [0.1,0.2,1] [-0.0,-1.29,-0.59] 5.0
operation_return: True
```
この状態を発見しようとすると，端末出力を監視しなければならなくなりそうですが・・・

ishiguroJSK, 2019-03-05:
std::coutにNULLを送るとコアダンプしたり表示が出なくなるのは既知らしく，
https://stackoverflow.com/questions/7019454/why-does-stdcout-output-disappear-completely-after-null-is-sent-to-it
コアダンプしない場合にもstd::cout.good()とかでエラーが立っているか検知は出来るみたいですが，
Pythonのテストスクリプトに実装(sys.stdout.errorsが該当？)してもプロセスやスレッドが違うと意味なさそうですし，
cppのコード内に埋め込むにも，随所にif(std::cout.good())を書くハメになるので不適な気がします．

```
  const char* s = NULL;
  std::cerr<<"[cerr] cout is alive?="<<std::cout.good()<<std::endl;
  std::cout<<"[cout] s="<<s<<std::endl;
  std::cout<<"[cout] after cout gets NULL"<<std::endl;
  std::cerr<<"[cerr] cout is alive?="<<std::cout.good()<<std::endl;
```
```
[cerr] cout is alive?=1
[cout] s=[cerr] cout is alive?=0
```
ちなみに上記リンクでの結論もstd::string使うか，std::coutに送る前にチェックするか，のようなのでテストで検出は無理なのではないでしょうか？

hsnuhayato, 2019-03-06:
> ちなみに上記リンクでの結論もstd::string使うか，std::coutに送る前にチェックするか，のようなのでテストで検出は無理なのではないでしょうか？

そうですね、調べてわかったこと:
1. streaming に null pointer渡すこと自体がundefined behaviorで、その後なにが起こるかはコンパイラ次第。
たまたまQCCがSIGSEGVをraiseしたらしい。
2. pythonから別プロセスのostreamの参照を取得することが無理っぽいです。そもそもpythonはCで実装され、c++のstream にアクセスする手段がない。pythonの sys.stdoutも標準出力をファイルディスクリプタで扱うラッパーで、c++とは違う実装。

> cppのコード内に埋め込むにも，随所にif(std::cout.good())を書くハメになるので不適な気がします．

同意見です。
シンプルにNULLをstreamに渡すのを避けましょう。というのが良いと思いますが...

k-okada, 2019-03-06:
@ishiguroJSK @hsnuhayato 丁寧な調査ありがとうございます．#1267 でOKだとおもいます．

closed 2019-03-28 by hsnuhayato