fkanehiro/hrpsys-base #1278: Issue in installation

mohamadmahdavian, 2019-07-11:
Hello

I get following error while making the package. Do you have any idea what might be the problem here?
[  0%] Built target hrpIo
[  0%] Built target Joystick
[  1%] Built target JoystickComp
[  2%] Built target Joystick2Velocity2D
[ 20%] Built target hrpsysBaseStub
[ 20%] Built target Joystick2Velocity3DComp
[ 20%] Built target Joystick2Velocity2DComp
[ 21%] Built target Joystick2Velocity3D
[ 22%] Built target Joystick2PanTiltAngles
[ 22%] Built target Joystick2PanTiltAnglesComp
[ 22%] Built target ForwardKinematicsComp
[ 23%] Built target ForwardKinematics
[ 24%] Built target HGcontrollerComp
[ 24%] Built target HGcontroller
[ 25%] Built target Range2PointCloud
[ 25%] Built target Range2PointCloudComp
[ 26%] Built target ImpedanceController
[ 27%] Built target JointPathExC
[ 28%] Built target testImpedanceOutputGenerator
[ 29%] Built target ImpedanceControllerComp
[ 30%] Built target testPreviewController
[ 32%] Built target testGaitGenerator
[ 34%] Built target AutoBalancer
[ 35%] Built target SoftErrorLimiter
[ 36%] Built target SoftErrorLimiterComp
[ 38%] Built target AutoBalancerComp
[ 39%] Built target VirtualForceSensorComp
[ 40%] Built target VirtualForceSensor
[ 41%] Built target GraspControllerComp
[ 42%] Built target GraspController
[ 42%] Built target TorqueFilterComp
[ 43%] Built target TorqueFilter
[ 43%] Built target testIIRFilter
[ 43%] Built target KalmanFilterComp
[ 44%] Built target KalmanFilter
[ 44%] Built target testKalmanFilterEstimation
[ 44%] Built target testKFilter
[ 45%] Built target Stabilizer
[ 46%] Built target testZMPDistributor
[ 47%] Built target testTwoDofController
[ 49%] Built target StabilizerComp
[ 50%] Built target RemoveForceSensorLinkOffset
[ 50%] Built target ServoController
[ 51%] Built target RemoveForceSensorLinkOffsetComp
[ 51%] Built target testServoSerial
[ 52%] Built target ServoControllerComp
[ 52%] Built target ThermoEstimator
[ 53%] Built target ThermoEstimatorComp
[ 54%] Built target ThermoLimiter
[ 54%] Built target ThermoLimiterComp
[ 56%] Built target TorqueController
[ 58%] Built target TorqueControllerComp
[ 58%] Building CXX object rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImage.dir/ImageData2CameraImage.o
[ 60%] Built target testMotorTorqueController
[ 60%] Building CXX object rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImageComp.dir/ImageData2CameraImage.o
[ 60%] Built target ExtractCameraImage
[ 61%] Built target ExtractCameraImageComp
[ 62%] Built target CaptureControllerComp
[ 63%] Built target CaptureController
[ 64%] Built target RangeNoiseMixer
Scanning dependencies of target AverageFilterComp
[ 64%] Built target RangeNoiseMixerComp
Scanning dependencies of target AverageFilter
[ 64%] Building CXX object rtc/AverageFilter/CMakeFiles/AverageFilter.dir/AverageFilter.o
[ 64%] Building CXX object rtc/AverageFilter/CMakeFiles/AverageFilterComp.dir/AverageFilterComp.o
In file included from /usr/include/openrtm-1.2/rtm/InPort.h:39:0,
                 from /usr/include/openrtm-1.2/rtm/DataInPort.h:23,
                 from /home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/ImageData2CameraImage/ImageData2CameraImage.h:18,
                 from /home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/ImageData2CameraImage/ImageData2CameraImage.cpp:10:
/usr/include/openrtm-1.2/rtm/Timestamp.h: In instantiation of ‘RTC::Timestamp<DataType>::ReturnCode RTC::Timestamp<DataType>::operator()(RTC::ConnectorInfo&, DataType&) [with DataType = Img::ImageData; RTC::Timestamp<DataType>::ReturnCode = RTC::ConnectorListenerStatus::Enum]’:
/home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/ImageData2CameraImage/ImageData2CameraImage.cpp:168:2:   required from here
/usr/include/openrtm-1.2/rtm/Timestamp.h:40:20: error: ‘struct Img::ImageData’ has no member named ‘tm’
       data.tm.sec  = tm.sec();
                    ^
/usr/include/openrtm-1.2/rtm/Timestamp.h:41:20: error: ‘struct Img::ImageData’ has no member named ‘tm’
       data.tm.nsec = tm.usec() * 1000;
                    ^
In file included from /usr/include/openrtm-1.2/rtm/InPort.h:39:0,
                 from /usr/include/openrtm-1.2/rtm/DataInPort.h:23,
                 from /home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/ImageData2CameraImage/ImageData2CameraImage.h:18,
                 from /home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/ImageData2CameraImage/ImageData2CameraImage.cpp:10:
/usr/include/openrtm-1.2/rtm/Timestamp.h: In instantiation of ‘RTC::Timestamp<DataType>::ReturnCode RTC::Timestamp<DataType>::operator()(RTC::ConnectorInfo&, DataType&) [with DataType = Img::ImageData; RTC::Timestamp<DataType>::ReturnCode = RTC::ConnectorListenerStatus::Enum]’:
/home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/ImageData2CameraImage/ImageData2CameraImage.cpp:168:2:   required from here
/usr/include/openrtm-1.2/rtm/Timestamp.h:40:20: error: ‘struct Img::ImageData’ has no member named ‘tm’
       data.tm.sec  = tm.sec();
                    ^
/usr/include/openrtm-1.2/rtm/Timestamp.h:41:20: error: ‘struct Img::ImageData’ has no member named ‘tm’
       data.tm.nsec = tm.usec() * 1000;
                    ^
rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImage.dir/build.make:62: recipe for target 'rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImage.dir/ImageData2CameraImage.o' failed
make[2]: *** [rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImage.dir/ImageData2CameraImage.o] Error 1
CMakeFiles/Makefile2:3476: recipe for target 'rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImage.dir/all' failed
make[1]: *** [rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImage.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 65%] Building CXX object rtc/AverageFilter/CMakeFiles/AverageFilterComp.dir/AverageFilter.o
rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImageComp.dir/build.make:75: recipe for target 'rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImageComp.dir/ImageData2CameraImage.o' failed
make[2]: *** [rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImageComp.dir/ImageData2CameraImage.o] Error 1
CMakeFiles/Makefile2:3513: recipe for target 'rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImageComp.dir/all' failed
make[1]: *** [rtc/ImageData2CameraImage/CMakeFiles/ImageData2CameraImageComp.dir/all] Error 2
[ 65%] Linking CXX shared library ../../lib/AverageFilter.so
[ 65%] Built target AverageFilter
[ 65%] Linking CXX executable ../../bin/AverageFilterComp
[ 65%] Built target AverageFilterComp
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

Thank You

fkanehiro, 2019-07-11:
You seems to be using OpenRTM-1.2 but I have never tried it.

mohamadmahdavian, 2019-07-11:
Can I know which version have you used for ubuntu 16.04?

Thank you

fkanehiro, 2019-07-11:
I'm using OpenRTM-1.1.2.

mohamadmahdavian, 2019-07-11:
Ok. Thank you

mohamadmahdavian, 2019-07-11:
Is it possible for you to send me the link for OpenRTM-1.1.2 installation. I have spend a long time for finding it,but I wasn't able to find 1.1.2 version.

Thank You

fkanehiro, 2019-07-11:
Please visit:
https://www.openrtm.org/openrtm/node/6040

mohamadmahdavian, 2019-07-12:
Thank You for your help. I was able to install openrtm-1.1.2. But I still have an error while making the package.Do you have any idea about this one?

Scanning dependencies of target JoystickComp
Scanning dependencies of target hrpIo
Scanning dependencies of target Joystick
[  0%] Generating PCDLoaderService.hh, PCDLoaderServiceSk.cpp, PCDLoaderServiceDynSk.cpp
[  0%] Building CXX object rtc/Joystick/CMakeFiles/JoystickComp.dir/JoystickComp.o
[  0%] Building CXX object rtc/Joystick/CMakeFiles/Joystick.dir/Joystick.o
[  0%] Building CXX object lib/io/CMakeFiles/hrpIo.dir/iob.o
In file included from /home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/Joystick/Joystick.cpp:10:0:
/home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/Joystick/Joystick.h:13:36: fatal error: rtm/idl/BasicDataType.hh: No such file or directory
compilation terminated.
/home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/Joystick/JoystickComp.cpp:10:25: fatal error: rtm/Manager.h: No such file or directory
compilation terminated.
rtc/Joystick/CMakeFiles/Joystick.dir/build.make:62: recipe for target 'rtc/Joystick/CMakeFiles/Joystick.dir/Joystick.o' failed
make[2]: *** [rtc/Joystick/CMakeFiles/Joystick.dir/Joystick.o] Error 1
CMakeFiles/Makefile2:1204: recipe for target 'rtc/Joystick/CMakeFiles/Joystick.dir/all' failed
make[1]: *** [rtc/Joystick/CMakeFiles/Joystick.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
rtc/Joystick/CMakeFiles/JoystickComp.dir/build.make:62: recipe for target 'rtc/Joystick/CMakeFiles/JoystickComp.dir/JoystickComp.o' failed
make[2]: *** [rtc/Joystick/CMakeFiles/JoystickComp.dir/JoystickComp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  1%] Building CXX object rtc/Joystick/CMakeFiles/JoystickComp.dir/Joystick.o
[  1%] Generating DataLoggerService.hh, DataLoggerServiceSk.cpp, DataLoggerServiceDynSk.cpp
In file included from /home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/Joystick/Joystick.cpp:10:0:
/home/mohammad/catkin_ws/lipm/hrpsys-base/rtc/Joystick/Joystick.h:13:36: fatal error: rtm/idl/BasicDataType.hh: No such file or directory
compilation terminated.
rtc/Joystick/CMakeFiles/JoystickComp.dir/build.make:75: recipe for target 'rtc/Joystick/CMakeFiles/JoystickComp.dir/Joystick.o' failed
make[2]: *** [rtc/Joystick/CMakeFiles/JoystickComp.dir/Joystick.o] Error 1
CMakeFiles/Makefile2:1167: recipe for target 'rtc/Joystick/CMakeFiles/JoystickComp.dir/all' failed
make[1]: *** [rtc/Joystick/CMakeFiles/JoystickComp.dir/all] Error 2
[  1%] Generating StateHolderService.hh, StateHolderServiceSk.cpp, StateHolderServiceDynSk.cpp
[  1%] Generating RobotHardwareService.hh, RobotHardwareServiceSk.cpp, RobotHardwareServiceDynSk.cpp
[  2%] Generating WavPlayerService.hh, WavPlayerServiceSk.cpp, WavPlayerServiceDynSk.cpp
[  2%] Generating SequencePlayerService.hh, SequencePlayerServiceSk.cpp, SequencePlayerServiceDynSk.cpp
[  2%] Generating ForwardKinematicsService.hh, ForwardKinematicsServiceSk.cpp, ForwardKinematicsServiceDynSk.cpp
[  2%] Generating ExecutionProfileService.hh, ExecutionProfileServiceSk.cpp, ExecutionProfileServiceDynSk.cpp
[  2%] Generating OGMap3DService.hh, OGMap3DServiceSk.cpp, OGMap3DServiceDynSk.cpp
[  3%] Generating NullService.hh, NullServiceSk.cpp, NullServiceDynSk.cpp
[  3%] Generating EmergencyStopperService.hh, EmergencyStopperServiceSk.cpp, EmergencyStopperServiceDynSk.cpp
[  3%] Generating TimeKeeperService.hh, TimeKeeperServiceSk.cpp, TimeKeeperServiceDynSk.cpp
[  3%] Generating Img.hh, ImgSk.cpp, ImgDynSk.cpp
[  3%] Generating HRPDataTypes.hh, HRPDataTypesSk.cpp, HRPDataTypesDynSk.cpp
[  4%] Generating pointcloud.hh, pointcloudSk.cpp, pointcloudDynSk.cpp
[  4%] Generating ImpedanceControllerService.hh, ImpedanceControllerServiceSk.cpp, ImpedanceControllerServiceDynSk.cpp
[  4%] Generating SoftErrorLimiterService.hh, SoftErrorLimiterServiceSk.cpp, SoftErrorLimiterServiceDynSk.cpp
[  4%] Generating CollisionDetectorService.hh, CollisionDetectorServiceSk.cpp, CollisionDetectorServiceDynSk.cpp
[  5%] Generating GraspControllerService.hh, GraspControllerServiceSk.cpp, GraspControllerServiceDynSk.cpp
[  5%] Generating TorqueFilterService.hh, TorqueFilterServiceSk.cpp, TorqueFilterServiceDynSk.cpp
[  5%] Generating KalmanFilterService.hh, KalmanFilterServiceSk.cpp, KalmanFilterServiceDynSk.cpp
[  5%] Generating VirtualForceSensorService.hh, VirtualForceSensorServiceSk.cpp, VirtualForceSensorServiceDynSk.cpp
[  5%] Generating StabilizerService.hh, StabilizerServiceSk.cpp, StabilizerServiceDynSk.cpp
[  5%] Generating AutoBalancerService.hh, AutoBalancerServiceSk.cpp, AutoBalancerServiceDynSk.cpp
[  6%] Generating RemoveForceSensorLinkOffsetService.hh, RemoveForceSensorLinkOffsetServiceSk.cpp, RemoveForceSensorLinkOffsetServiceDynSk.cpp
[  6%] Generating ServoControllerService.hh, ServoControllerServiceSk.cpp, ServoControllerServiceDynSk.cpp
[  6%] Generating TorqueControllerService.hh, TorqueControllerServiceSk.cpp, TorqueControllerServiceDynSk.cpp
[  6%] Generating ThermoLimiterService.hh, ThermoLimiterServiceSk.cpp, ThermoLimiterServiceDynSk.cpp
[  6%] Generating ReferenceForceUpdaterService.hh, ReferenceForceUpdaterServiceSk.cpp, ReferenceForceUpdaterServiceDynSk.cpp
[  6%] Generating AccelerationFilterService.hh, AccelerationFilterServiceSk.cpp, AccelerationFilterServiceDynSk.cpp
[  7%] Generating ObjectContactTurnaroundDetectorService.hh, ObjectContactTurnaroundDetectorServiceSk.cpp, ObjectContactTurnaroundDetectorServiceDynSk.cpp
Scanning dependencies of target hrpsysBaseStub
[... cut ...]
  Policy CMP0081 is not set: Relative paths not allowed in LINK_DIRECTORIES
  target property.  Run "cmake --help-policy CMP0081" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Found relative path while evaluating link directories of "RobotHardware":

    "io"

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/mohammad/catkin_ws/lipm/hrpsys-base/build

Thank You

mohamadmahdavian, 2019-07-12:
I was able to fix that problem.Now I am getting this error.Do you how I can solve it?

/////////////////////////////
[  0%] Generating PCDLoaderService.hh, PCDLoaderServiceSk.cpp, PCDLoaderServiceDynSk.cpp
[  0%] Building CXX object rtc/Joystick/CMakeFiles/Joystick.dir/Joystick.o
[  0%] Built target hrpIo
[  0%] Building CXX object rtc/Joystick/CMakeFiles/JoystickComp.dir/JoystickComp.o
[  1%] Building CXX object rtc/Joystick/CMakeFiles/JoystickComp.dir/Joystick.o
/home/mohammad/catkin_ws/lipm/hrpsys-base/idl/PCDLoaderService.idl:5: ExtendedDataTypes.idl: No such file or directory
/home/mohammad/catkin_ws/lipm/hrpsys-base/idl/PCDLoaderService.idl:12: Error in look-up of 'RTC::Pose3D': 'RTC' not found
/home/mohammad/catkin_ws/lipm/hrpsys-base/idl/PCDLoaderService.idl:13: Error in look-up of 'RTC::Point3D': 'RTC' not found
omniidl: 2 errors.
omniidl: Error running preprocessor
idl/CMakeFiles/hrpsysBaseStub.dir/build.make:361: recipe for target 'idl/PCDLoaderService.hh' failed
make[2]: *** [idl/PCDLoaderService.hh] Error 1
CMakeFiles/Makefile2:108: recipe for target 'idl/CMakeFiles/hrpsysBaseStub.dir/all' failed
make[1]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Building CXX object rtc/Joystick/CMakeFiles/JoystickComp.dir/js.o
[  1%] Linking CXX executable ../../bin/JoystickComp
[  1%] Linking CXX shared library ../../lib/Joystick.so
[  1%] Built target JoystickComp
[  1%] Built target Joystick
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

Thank You

closed 2019-10-24 by mohamadmahdavian

Aimiliospet, 2022-07-21:
@fkanehiro @ghost did you manage to solve the issue? I'm currently facing the last error with ubuntu 18.04 .
when i use make i get:
`[  0%] Generating PCDLoaderService.hh, PCDLoaderServiceSk.cpp, PCDLoaderServiceDynSk.cpp
omniidl: No files specified. Use 'omniidl -u' for usage.
idl/CMakeFiles/hrpsysBaseStub.dir/build.make:361: recipe for target 'idl/PCDLoaderService.hh' failed
make[2]: *** [idl/PCDLoaderService.hh] Error 1
CMakeFiles/Makefile2:103: recipe for target 'idl/CMakeFiles/hrpsysBaseStub.dir/all' failed
make[1]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
`