fkanehiro/hrpsys-base #1279: Is ubuntu 16 supported

mohamadmahdavian, 2019-07-12:
Hello

You have mentioned in:"http://fkanehiro.github.io/hrpsys-base/"  that ubuntu 12 and 14 are supported.I am not sure how old is that link. Is it possible to use this package in ubuntu 16.04 currently?

Thank You

ishiguroJSK, 2019-07-14:
We can run in ubuntu 16.04 now.
With only this package, you can check like this. (I hope...)
```
~/catkin_ws/debug/src $ git clone https://github.com/fkanehiro/hrpsys-base.git
~/catkin_ws/debug/src $ rosdep install -y -r --from-paths . --ignore-src
~/catkin_ws/debug/src $ catkin b
~/catkin_ws/debug/src/hrpsys-base/launch $ roslaunch ./samplerobot.launch
~/catkin_ws/debug/src/hrpsys-base/launch $ rosrun hrpsys samplerobot-walk.py
```

mohamadmahdavian, 2019-07-15:
Thank you for your respond. I get following error while launching file.Do you have any idea what might be the problem?

... logging to /home/mohammad/.ros/log/78673210-a723-11e9-8e9f-ec0ec422416d/roslaunch-mohammad-28533.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Invalid <arg> tag: hrpsys
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share. 

Arg xml is <arg default="$(find hrpsys)/samples/SampleRobot/SampleRobot.$(arg CONTROLLER_PERIOD).conf" name="CONF_FILE"/>
The traceback for the exception was written to the log file

ishiguroJSK, 2019-07-16:
I guess it seems to be missing PATH problem.
Sorry, I forgot to tell you about `source ○○/devel/setup.bash`.
When you create ROS catkin workspace, don't forget to setup path by `source ~/○○/devel/setup.bash`
(Most people write it on~/.bashrc)
http://wiki.ros.org/kinetic/Installation/Ubuntu#kinetic.2BAC8-Installation.2BAC8-DebEnvironment.Environment_setup