fkanehiro/hrpsys-base #1291: make時にBasicDataType.hhが見つからないというエラーが出てしまう

HidemiANDO, 2020-06-22:
hrpsys-baseをmakeしようとした際

In file included from /home/ando-h/hrpsys-base-master/idl/RobotHardwareServiceSk.cpp:3:0:
/home/ando-h/hrpsys-base-master/idl/RobotHardwareService.hh:23:28: fatal error: BasicDataType.hh: そのようなファイルやディレクトリはありません
compilation terminated.
idl/CMakeFiles/hrpsysBaseStub.dir/build.make:468: ターゲット 'idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o' のレシピで失敗しました
make[2]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o] エラー 1
CMakeFiles/Makefile2:103: ターゲット 'idl/CMakeFiles/hrpsysBaseStub.dir/all' のレシピで失敗しました
make[1]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/all] エラー 2
Makefile:160: ターゲット 'all' のレシピで失敗しました
make: *** [all] エラー 2

のようなエラーが出てしまいます。BasicDataType.hhがOpenRTMをインストールすることによってインストールされるということはわかっていて、OpenRTMの1.1.2をインストール済みでcmake時に
-- Found OpenRTM-aist 1.1.2 in /usr/local
と表示されているのでパスは通っているのではないかと思われるため、このようなエラーが出てしまう理由がよくわかっていません。issuesを確認すると以前同じようなエラーが出た方がおられるようなのですが、解決法までは記載されていらっしゃらなかったので、ご教授いただけると幸いです。

fkanehiro, 2020-06-23:
- OSは何をお使いでしょうか
- BasicDataType.hhはどこにインストールされているでしょうか
- `make VERBOSE=1`とした時にBasicDataType.hhのあるパスにインクルードパスが通っているでしょうか

上記3点を確認してみて下さい。

HidemiANDO, 2020-06-23:
お忙しいところお返事ありがとう御座います。
また、お伝えするのを忘れてしまっていたのですが、自分の環境ではインストールされる場所がusr/ではなくusr/local/となっていたので/hrpsys-base-master/cmake_modulesのFindOpenRTM.cmakeの１１行目
set(OPENRTM_CONFIG_EXECUTABLE ${OPENRTM_DIR}/bin/rtm-config)
をset(OPENRTM_CONFIG_EXECUTABLE ${OPENRTM_DIR}/local/bin/rtm-config)
と書き換えています。

・OSはUbuntu16.04を使用しています。
・BasicDataType.hhは/usr/local/include/openrtm-1.1/rtm/idlの下にインストールされていました。
・make VERBOSE=1とした際の出力は

$ make VERBOSE=1
/usr/bin/cmake -H/home/ando-h/hrpsys-base-master -B/home/ando-h/hrpsys-base-master --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/ando-h/hrpsys-base-master/CMakeFiles /home/ando-h/hrpsys-base-master/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: ディレクトリ '/home/ando-h/hrpsys-base-master' に入ります
make -f idl/CMakeFiles/hrpsysBaseStub.dir/build.make idl/CMakeFiles/hrpsysBaseStub.dir/depend
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master' に入ります
cd /home/ando-h/hrpsys-base-master && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ando-h/hrpsys-base-master /home/ando-h/hrpsys-base-master/idl /home/ando-h/hrpsys-base-master /home/ando-h/hrpsys-base-master/idl /home/ando-h/hrpsys-base-master/idl/CMakeFiles/hrpsysBaseStub.dir/DependInfo.cmake --color=
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master' から出ます
make -f idl/CMakeFiles/hrpsysBaseStub.dir/build.make idl/CMakeFiles/hrpsysBaseStub.dir/build
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master' に入ります
[  1%] Building CXX object idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o
cd /home/ando-h/hrpsys-base-master/idl && /usr/bin/c++   -DDISABLE_OPENNI2 -DDISABLE_PCAP -DDISABLE_PNG -DROBOT_IOB_VERSION=4 -DhrpsysBaseStub_EXPORTS -DvtkFiltersFlowPaths_AUTOINIT="1(vtkFiltersParallelFlowPaths)" -DvtkIOExodus_AUTOINIT="1(vtkIOParallelExodus)" -DvtkIOGeometry_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOImage_AUTOINIT="1(vtkIOMPIImage)" -DvtkIOSQL_AUTOINIT="2(vtkIOMySQL,vtkIOPostgreSQL)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL)" -DvtkRenderingCore_AUTOINIT="4(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingFreeTypeOpenGL,vtkRenderingOpenGL)" -DvtkRenderingFreeType_AUTOINIT="2(vtkRenderingFreeTypeFontConfig,vtkRenderingMatplotlib)" -DvtkRenderingLIC_AUTOINIT="1(vtkRenderingParallelLIC)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL)" -I/home/ando-h/hrpsys-base-master -I/usr/local/include/OpenHRP-3.1 -I/usr/include/eigen3 -I/usr/local/include -I/usr/local/include/rtm/idl -I/usr/include/vtk-6.2 -I/usr/include/python2.7 -I/usr/include/x86_64-linux-gnu -I/usr/include/freetype2 -I/usr/include/x86_64-linux-gnu/freetype2 -I/usr/include/hdf5/openmpi -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include -I/usr/lib/openmpi/include -I/usr/lib/openmpi/include/openmpi -I/usr/include/libxml2 -I/usr/include/jsoncpp -I/usr/include/tcl -I/usr/include/opencv  -O3 -DNDEBUG -fPIC   -DHRPSYS_PACKAGE_VERSION="\"315.15.0\"" -o CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o -c /home/ando-h/hrpsys-base-master/idl/RobotHardwareServiceSk.cpp
In file included from /home/ando-h/hrpsys-base-master/idl/RobotHardwareServiceSk.cpp:3:0:
/home/ando-h/hrpsys-base-master/idl/RobotHardwareService.hh:23:28: fatal error: BasicDataType.hh: そのようなファイルやディレクトリはありません
compilation terminated.
idl/CMakeFiles/hrpsysBaseStub.dir/build.make:468: ターゲット 'idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o' のレシピで失敗しました
make[2]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o] エラー 1
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master' から出ます
CMakeFiles/Makefile2:103: ターゲット 'idl/CMakeFiles/hrpsysBaseStub.dir/all' のレシピで失敗しました
make[1]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/all] エラー 2
make[1]: ディレクトリ '/home/ando-h/hrpsys-base-master' から出ます
Makefile:160: ターゲット 'all' のレシピで失敗しました
make: *** [all] エラー 2

となっており、真ん中辺りの-I/usr/local/include/rtm/idlがインクルードパスであれば、実はパスの指定がうまく行っていないのかもしれません。

また、インクルードパスの設定について、/hrpsys-base-master/cmake_modulesのFindOpenRTM.cmakeの4１行目から53行目の
if((NOT (OPENRTM_VERSION VERSION_LESS "1.1.0")) AND (OPENRTM_VERSION VERSION_LESS "1.2.0"))
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/openrtm-1.1/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.1")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.1/rtm/idl")
        elseif(NOT (OPENRTM_VERSION VERSION_LESS "1.2.0"))
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/openrtm-1.2/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.2")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.2/rtm/idl")
	else()
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/rtm/idl")
	endif()
であっていますでしょうか。
こちらのすべての${OPENRTM_DIR}と/includeの間に/localを追加してみたり、バージョンがうまく認識されていない可能性を考慮し下部の
	else()
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/rtm/idl")
	endif()
の箇所を
	else()
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/openrtm-1.1/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.1")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.1/rtm/idl")
	endif()
と書き換えたりしてみたのですが、
make VERBOSE=1の実行結果の出力は上記に示したものと変化が見られませんでした。

fkanehiro, 2020-06-23:
```
-- Found OpenRTM-aist 1.1.2 in /usr/local
```
と出ている（OPENRTM_DIRとして/usr/localが検出された）とのことなので、
```
/hrpsys-base-master/cmake_modulesのFindOpenRTM.cmakeの１１行目
set(OPENRTM_CONFIG_EXECUTABLE ${OPENRTM_DIR}/bin/rtm-config)
をset(OPENRTM_CONFIG_EXECUTABLE ${OPENRTM_DIR}/local/bin/rtm-config)
と書き換えています。
```
は不要ではないでしょうか？

HidemiANDO, 2020-06-23:
返信有り難うございます。

その箇所についてなのですが、
set(OPENRTM_CONFIG_EXECUTABLE ${OPENRTM_DIR}/bin/rtm-config)
のままだと、cmake ..を実行した際に

$ cmake ..
CMake Error at cmake_modules/FindOpenRTM.cmake:14 (message):
  rtm-config was not found in /usr/bin.  Please set OPENRTM_DIR correctly.
Call Stack (most recent call first):
  CMakeLists.txt:61 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ando-h/hrpsys-base-master/CMakeFiles/CMakeOutput.log".
See also "/home/ando-h/hrpsys-base-master/CMakeFiles/CMakeError.log".

というエラーが発生してしまい、cmake ..自体が通らなくなってしまっています。

fkanehiro, 2020-06-23:
その部分を通るということは誰かがOPENRTM_DIRを設定しているということですが、誰が設定しているのでしょうか？

HidemiANDO, 2020-06-23:
申し訳ありません。確認してみたところ、CMakeCache.txtにおいて
//The top directory of OpenRTM-aist
OPENRTM_DIR:PATH=/usr
となっていたため
CMakeCache.txtを一度削除してcmake ..を再試行したところ、cmake ..が通りました。
しかし、やはりmakeを実行するとやはりBasicDataType.hhが見つからないというエラーが発生してしまいました。make VERBOSE＝１を実行すると先ほど（https://github.com/fkanehiro/hrpsys-base/issues/1291#issuecomment-647919209）と同様の出力になってしまいます。

fkanehiro, 2020-06-23:
OPENRTM_VERSIONがどう検出されているかご確認いただけないでしょうか。

HidemiANDO, 2020-06-23:
返信有り難うございます。
OPENRTM_VERSIONの検出プロセスについて、よくわかっていませんでした。申し訳ありません。
確認したところインストールされたOpenRTMのｒｔｍ-configのバージョンを--versionで取得し、その値がOPENRTM_VERSIONに入るのであっていますでしょうか。

同様に手動でコマンドラインからインストールされているOpenRTMのrtm-configのバージョンを確認すると1.1.2となっており、また、cmake ..実行時にFindOpenRTM.cmakeの
if((NOT (OPENRTM_VERSION VERSION_LESS "1.1.0")) AND (OPENRTM_VERSION VERSION_LESS "1.2.0"))
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/openrtm-1.1/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.1")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.1/rtm/idl")
        elseif(NOT (OPENRTM_VERSION VERSION_LESS "1.2.0"))
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/openrtm-1.2/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.2")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/openrtm-1.2/rtm/idl")
	else()
	  set(OPENRTM_IDL_DIR ${OPENRTM_DIR}/include/rtm/idl)
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include")
	  list(APPEND OPENRTM_INCLUDE_DIRS "${OPENRTM_DIR}/include/rtm/idl")
	endif()
においてmessageを出力したところ、最初の条件分岐にしっかり入っていることが確認出来たので、
OPENRTM_VERSIONには正しく1.1.2の値が入っているのではないかと思われます。

fkanehiro, 2020-06-23:
おかしいですね。そうすると正しいインストールパスが設定されるはずですが。
`make VERBOSE=1`の出力を確認してみてください。

HidemiANDO, 2020-06-27:
お忙しいところご返信頂きありがとうございます。
長くなってしまい大変申し訳ありません。make VERBOSE=1の出力結果は以下のようになっておりまして、
************************************************************************************************:
$ make VERBOSE=1
/usr/bin/cmake -H/home/ando-h/hrpsys-base-master -B/home/ando-h/hrpsys-base-master/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/ando-h/hrpsys-base-master/build/CMakeFiles /home/ando-h/hrpsys-base-master/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: ディレクトリ '/home/ando-h/hrpsys-base-master/build' に入ります
make -f idl/CMakeFiles/hrpsysBaseStub.dir/build.make idl/CMakeFiles/hrpsysBaseStub.dir/depend
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master/build' に入ります
cd /home/ando-h/hrpsys-base-master/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ando-h/hrpsys-base-master /home/ando-h/hrpsys-base-master/idl /home/ando-h/hrpsys-base-master/build /home/ando-h/hrpsys-base-master/build/idl /home/ando-h/hrpsys-base-master/build/idl/CMakeFiles/hrpsysBaseStub.dir/DependInfo.cmake --color=
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master/build' から出ます
make -f idl/CMakeFiles/hrpsysBaseStub.dir/build.make idl/CMakeFiles/hrpsysBaseStub.dir/build
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master/build' に入ります
[  0%] Building CXX object idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o
cd /home/ando-h/hrpsys-base-master/build/idl && /usr/bin/c++   -DDISABLE_OPENNI2 -DDISABLE_PCAP -DDISABLE_PNG -DROBOT_IOB_VERSION=4 -DhrpsysBaseStub_EXPORTS -DvtkFiltersFlowPaths_AUTOINIT="1(vtkFiltersParallelFlowPaths)" -DvtkIOExodus_AUTOINIT="1(vtkIOParallelExodus)" -DvtkIOGeometry_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOImage_AUTOINIT="1(vtkIOMPIImage)" -DvtkIOSQL_AUTOINIT="2(vtkIOMySQL,vtkIOPostgreSQL)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL)" -DvtkRenderingCore_AUTOINIT="4(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingFreeTypeOpenGL,vtkRenderingOpenGL)" -DvtkRenderingFreeType_AUTOINIT="2(vtkRenderingFreeTypeFontConfig,vtkRenderingMatplotlib)" -DvtkRenderingLIC_AUTOINIT="1(vtkRenderingParallelLIC)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL)" -I/home/ando-h/hrpsys-base-master/build -I/usr/local/include/OpenHRP-3.1 -I/usr/include/eigen3 -I/usr/local/include -I/usr/local/include/rtm/idl -I/usr/include/vtk-6.2 -I/usr/include/python2.7 -I/usr/include/x86_64-linux-gnu -I/usr/include/freetype2 -I/usr/include/x86_64-linux-gnu/freetype2 -I/usr/include/hdf5/openmpi -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include -I/usr/lib/openmpi/include -I/usr/lib/openmpi/include/openmpi -I/usr/include/libxml2 -I/usr/include/jsoncpp -I/usr/include/tcl -I/usr/include/opencv  -O3 -DNDEBUG -fPIC   -DHRPSYS_PACKAGE_VERSION="\"315.15.0\"" -o CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o -c /home/ando-h/hrpsys-base-master/build/idl/RobotHardwareServiceSk.cpp
In file included from /home/ando-h/hrpsys-base-master/build/idl/RobotHardwareServiceSk.cpp:3:0:
/home/ando-h/hrpsys-base-master/build/idl/RobotHardwareService.hh:23:28: fatal error: BasicDataType.hh: そのようなファイルやディレクトリはありません
compilation terminated.
idl/CMakeFiles/hrpsysBaseStub.dir/build.make:468: ターゲット 'idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o' のレシピで失敗しました
make[2]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/RobotHardwareServiceSk.o] エラー 1
make[2]: ディレクトリ '/home/ando-h/hrpsys-base-master/build' から出ます
CMakeFiles/Makefile2:103: ターゲット 'idl/CMakeFiles/hrpsysBaseStub.dir/all' のレシピで失敗しました
make[1]: *** [idl/CMakeFiles/hrpsysBaseStub.dir/all] エラー 2
make[1]: ディレクトリ '/home/ando-h/hrpsys-base-master/build' から出ます
Makefile:149: ターゲット 'all' のレシピで失敗しました
make: *** [all] エラー 2
*********************************************************************************************************************************
出力の２５行目付近に-I/usr/local/include/rtm/idlとあることから、最終的に何らかの理由でインクルードパスがおかしくなってしまっているのでは無いかと思われます。

気になったことがありまして、一応INSTALLに従い依存するソフトウェアはインストールしているのですが、再び長くなってしまい非常に申し訳ないのですがcmake ..の出力が以下のようになっておりまして
********************************************************************************************************************************
$ cmake .
-- Found OpenRTM-aist 1.1.2 in /usr/local
-- Found OpenHRP 3.1.9 in /usr/local
-- OPENHRP_DEFINITIONS: 
-- REMOVE -Ox option from OPENHRP_DEFINITIONS, use CMAKE_BUILD_TYPE
-- OPENHRP_DEFINITIONS: 
-- Compile with Release
--  -- CMAKE_C_FLAGS / CMAKE_CXX_FLAGS  / 
--  -- CMAKE_C_FLAGS_DEBUG / CMAKE_CXX_FLAGS_DEBUG -g / -g
--  -- CMAKE_C_FLAGS_RELEASE / CMAKE_CXX_FLAGS_RELEASE -O3 -DNDEBUG / -O3 -DNDEBUG
--  -- CMAKE_C_FLAGS_RELWITHDEBINFO / CMAKE_CXX_FLAGS_RELWITHDEBINFO -O2 -g -DNDEBUG / -O2 -g -DNDEBUG
-- Found Qhull in /usr
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
[... cut ...]
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/ando-h/hrpsys-base-master/build

********************************************************************************************************************************
以上のようにいくつかソフトウェア関連の警告が出ております。これらの警告が出ているソフトウェアについては、オプションだと思っていたのですが、これらがインストールされていないことによってインクルードパスが正しく設定されていないという可能性はありますでしょうか。
また、Policy CMP00XX is not setというエラーも多く出ておりまして、自分の環境ではcmakeは3.5.1を使用しているのですが、cmakeのバージョンによってはmakeが通らなくなる可能性はありますでしょうか。

fkanehiro, 2020-06-29:
`pkg-config --cflags openhrp3.1`としてみてください。その出力に`/usr/local/include/rtm/idl`が含まれているようであれば、openhrrp3.1の方をまずは直す必要がありそうです。

HidemiANDO, 2020-06-30:
お忙しいところご返信ありがとうございます。
pkg-config --cflags openhrp3.1を実行してみたところ、/usr/local/include/rtm/idlが含まれていました。openHRPをインストールする際に、一括パッケージインストールスクリプトに含まれていたopenRTM1.0.0を手動で削除した後に、openHRP本体をインストールし直していなかったことが原因になっていたエラーだったようです。
その点に注意してopenHRPをインストールしなおしたところ、pkg-config --cflags openhrp3.1の出力に含まれるパスが
正しく/usr/local/include/openrtm-1.1/rtm/idlとなりました。
その後hrpsys-baseの方でcmake ..とmakeを実行したところ、正常にビルドまでまで出来た模様です。
最初の問題は解決したので、一度スレッドの方を閉じさせていただこうと思います。
再び何かしらの問題が発生したら恐縮ですが質問させていただけたら幸いです。
この度はお忙しい中問題解決にご協力いただき、誠にありがとうございました。

closed 2020-06-30 by HidemiANDO