fkanehiro/hrpsys-base #1292: hrpsys-baseと連携させるChoreonoidのバージョンについて

HidemiANDO, 2020-07-01:
不慣れな質問で申し訳ありません。
hrpsys-baseとchoreonoidを連携させて使用しようとした場合、連携させるchoreonoidのバージョンはいくつが望ましいでしょうか。

ishiguroJSK, 2020-07-01:
hrpsysとchoreonoidの連携はやや複雑で，各々の組織で独自のシステムを構成して運用している状態だと認識していますが，
稲葉岡田研(JSK)方式でよろしければ
https://github.com/start-jsk/rtmros_choreonoid
https://github.com/start-jsk/rtmros_choreonoid/wiki
あたりにインストール方法が書いてあります．

Ubuntu16だとchoreonoid1.6で，Ubuntu18だとchoreonoid1.7で実績があります．

ただこの場合，ROSやcatkinワークスペースの扱いに慣れておく必要と，
いくつかのレポジトリはprivateなので無視すること，
openrtmのパスなどを変えている場合，適宜読み替える必要があることが注意点だと思います．

先のissueでUbuntu16にopenrtm1.1.2を手動でインストールしているようですが，
単純にhrpsys->openrtm->choreonoidでのヒューマノイドシミュレーションを試したいということであれば，
できるだけapt-getで済ませる方が楽かと思います

そうではなく自力でhrpsysとchoreonoidのopenrtmポートを繋いで運用する場合も，1.6か1.7になると思います

fkanehiro, 2020-07-02:
こちらでは
https://github.com/isri-aist/choreonoid
を使用しています。

HidemiANDO, 2020-07-02:
@ishiguroJSK 
ご丁寧にお答えいただき、ありがとうございます。
また、プログラムまでご紹介頂き、ありがとうございます。
ROSについては多少ながら理解があるため、おそらく問題ないと思います。
教えていただいたプログラムの方を試させていただきたいと思うのですが、もし不明点が生じた場合、差し支えなければリポジトリのissuesの方でお聞きさせていただくことは可能でしょうか。

HidemiANDO, 2020-07-02:
@fkanehiro 
お忙しいところご返信頂き、ありがとうございます。
開発版について、自分の環境ではopenRTM分離後のものがopenRTMプラグインをONにするとchoreonoid単体でもビルドが通らなくなるエラーが出ているので、まずそちらの方を解決してから試してみようと思います。
貴重な情報をお教え頂き、ありがとうございます。

ishiguroJSK, 2020-07-02:
> 教えていただいたプログラムの方を試させていただきたいと思うのですが、もし不明点が生じた場合、差し支えなければリポジトリのissuesの方でお聞きさせていただくことは可能でしょうか。

はい．大丈夫です．

HidemiANDO, 2020-07-02:
ありがとうございます。
よろしくお願いいたします。

closed 2020-07-02 by HidemiANDO